cole8888/SRNE-Solar-Charge-Controller-Monitor #3: [QUESTION]

curiousIya, 2022-02-16:
Hi! Thank you so much for your python code :) I'm using it to read my ML4860 Solar Controller.
I removed the mqtt and the CC1 lines.

Just some questions, 
How do you know how many registers to read? 
And which registers refers to a value (eg r.registers[7] is Panel Voltage)?
Also, whose temperature do you refer in CPUTemperature().temperature? the raspberry pi?

cole8888, 2022-02-16:
You can determine which register corresponds to which value using the modbus manual I included here: https://github.com/cole8888/SRNE-Solar-Charge-Controller-Monitor/blob/main/srne_modbus_protocol-_3.8-english.pdf

For instance, at the start of the "Controller dynamic information" section you can see 0100H (100 in hexadecimal, 256 in decimal) is the battery state of charge percentage.

As for the number of registers, the reason I read 34 registers in my code is because that is the end of the "Controller dynamic information" section (122H) and beyond this address you start getting into the memory for the controller settings, which I was not interested in.

So if you only ever wanted to read the battery state of charge percentage and the battery voltage and nothing else, you could change that 34 to 2 since you'd only need to be reading registers 100H and 101H. If I were you I'd leave it at 34 though since that way it returns all the information you could possibly need.

Also, yes the CPU temperature was for the raspberry pi. The deck box I use can get pretty toasty and I wanted to make sure I wasn't cooking it :P

curiousIya, 2022-02-16:
Thank you very much! :)

To check if there is error, should I read r.registers[33] and r.registers[34]?

Also, do you know how to write register/enable load?

cole8888, 2022-02-17:
By error I assume you mean faults reported by the controller such as shorts and low battery warnings? 

After looking though the Arduino code again and the manual, I think you only have to read register 34 (aka 122H in the manual) since the error data is only stored in the higher bits and bits 0-15 are reserved. I'm pretty sure my python code is wrong since it reads register 33 instead of 34 (or 121H instead of 122H). In this case you may need to increase the 34 in
```r = modbus.read_holding_registers(256, 34, unit=1)``` to a 35 and use r.registers[34] to get the fault codes. Upon closer inspection it looks like I fixed this in the arduino code and forgot to update the python code with this fix.

As for extracting the fault codes from the data in this register, the raspberry pi or arduino code doesn't have any code for this because I handled it on the web interface viewer I made.

I had an array of fault strings (order and index is important!):
```
faultCodes[
    "Charge MOS short circuit",     //0
    "Anti-reverse MOS short",       //1
    "PV panel reversely connected", //2
    "PV working point over voltage",//3
    "PV counter current",           //4
    "PV input side over-voltage",   //5
    "PV input side short circuit",  //6
    "PV input overpower",           //7
    "Ambient temp too high",        //8
    "Controller temp too high",     //9
    "Load over-power/current",      //10
    "Load short circuit",           //11
    "Battery undervoltage warning", //12
    "Battery overvoltage",          //13
    "Battery over-discharge"];      //14
```
And here is the code I used for parsing the register data and making a string that contains the current errors:
```
  var faults = "None :)";
  if(faultID != 0){
      faults = "";
      var count = 0;
      while(faultID != 0){
          if(faultID >= Math.pow(2, 15-count)){
              // If there is more than one error, make a new line.
              if(count > 0){
                  faults += '<br>';
              }
              faults += '- ' + faultCodes[count-1];
              faultID -= Math.pow(2, 15-count);
          }
          count += 1;
      }
  }
```
(where faultID is the r.registers[34] data)
This is from a javascript program so you will need to adapt it to python for your use case. Let me know if you need any help with that. Essentially what I do is repeatedly check to see if I can subtract a power of two, and if I can then I know which bit it is and therefore which error(s) are present.

The deck box where I keep my charge controllers is not easily accessible during the winter so I cannot hook up my raspberry pi to test anything I've said here, so let me know if you encounter any issues.

closed 2022-11-25 by cole8888