cole8888/SRNE-Solar-Charge-Controller-Monitor #10: 1 Arduino only

Rey1635, 2022-11-20:
As a newbie, I would like to ask how to use only 1 arduino to read the battery voltage and the charging state status of the SRNE? How will I connect it to arduino? Thank you very much for your help.

cole8888, 2022-11-20:
Hi, sorry about not thinking about that, I'll make a schematic / code for a single arduino and charge controller.
Will you have access to an esp8266? I'm thinking i can get it to work with just one esp8266. That way you don't have to deal with the RF24 stuff.

Rey1635, 2022-11-20:
Thank you very much for your response.. I do have an esp8266. I'll just wait for your update.  God Bless You always.

cole8888, 2022-11-20:
You are also going to need a MAX3232 breakout board since the voltage level the charge controller uses to talk over rs232 isn't compatible with the Arduino and needs to be converted.

I found [this](https://www.amazon.com/Treedix-MAX3232-Convert-Transceiver-Breakout/dp/B09BCKWL8V) on amazon and it should work.

I'm not sure where you are from but try searching for "MAX3232 breakout" and see if you can find something simmilar to that, feel free to send links to the ones you find and I can try to tell you if they will work.

If you can only find ones which look like [this](https://www.amazon.com/NOYITO-Module-Conversion-Arduino-communicates/dp/B07BJJC3R6/) that's ok, you'll need to find a way to connect to the DB9 connector using something like [this](https://www.amazon.com/Connector-Female-Adapter-Terminal-Signal/dp/B086VYDPKH/).

cole8888, 2022-11-20:
I managed to get a ESP8266 to talk to the charge controller, will upload code, schematic and pictures shortly.

Rey1635, 2022-11-20:
I do have one of [this](https://www.amazon.com/HiLetgo-Level-Converter-Board-Rs232/dp/B00LUDCAXQ/ref=sr_1_2?crid=2SUUNSQ4SYUPG&keywords=Mini+RS232+to+TTL+chip+MAX3232&qid=1668986257&sprefix=mini+rs232+to+ttl+chip+max3232%2Caps%2C283&sr=8-2) and I already have a RJ12 (6P6C).  Thank you very much

cole8888, 2022-11-21:
Nice, that should work!

I uploaded some examples and much more detailed instructions, have a look at them [here](https://github.com/cole8888/SRNE-Solar-Charge-Controller-Monitor/tree/main/Arduino-Version).

Also [this particular example](https://github.com/cole8888/SRNE-Solar-Charge-Controller-Monitor/blob/main/Arduino-Version/Single-Charge-Controller/SOLAR_SINGLE_CC/SOLAR_SINGLE_CC.ino) is the one you are interested in, just make sure you replace the `MAX3232_RX` and `MAX3232_TX` values to whatever pins you are using.

Let me know if you run into any problems :)

Rey1635, 2022-11-21:
Thank you very much.  I'm so excited to start my project.  May God Bless You always.

Rey1635, 2022-11-21:
I tried to the upload code on my esp8266 but there is an error ("'class Modbus' has no member named 'start'") on master.start(); I've already  installed the modbus-esp8266 library.

cole8888, 2022-11-21:
I'm not using that modbus library, I used https://github.com/smarmengol/Modbus-Master-Slave-for-Arduino
There are instructions in that repoitory for how to install it, I don't think it is in the Arduino IDE.

Rey1635, 2022-11-21:
thanks for the link of the library.. error has been solved.

Rey1635, 2022-11-21:
Sad to say no luck for me.. I have already tried with arduino. 
[image]

Rey1635, 2022-11-21:
Maybe I should purchase a different type of  MAX3232 breakout board.

cole8888, 2022-11-21:
How do you have it wired up? You may have RX and TX backwards on either the arduino side or the charge controller side.

Rey1635, 2022-11-22:
Here is how  I wired up.
[image]

cole8888, 2022-11-22:
On the arduino side the you have the VCC and GND reversed. Also I'm not sure what roles you have assigned to D2 and D3. For getting it started for the first time, can you stick to D1 and D2 since those are the only pins I tested.
[image]
Your RX and TX lines seem correct though.

cole8888, 2022-11-22:
I'm pretty sure D3 cannot be used for software serial according to some stuff I found online. Just use D1 and D2 for now since I know those work.

Rey1635, 2022-11-22:
Thank you very much.

cole8888, 2022-11-22:
Were you able to get it to work?

I have Arduino nano wiring examples [here](https://github.com/cole8888/SRNE-Solar-Charge-Controller-Monitor/tree/main/Arduino-Version/Single-Charge-Controller/Photos%20and%20Schematics).
So, if you follow those examples use:
`#define MAX3232_RX 2`
`#define MAX3232_TX 3`

Rey1635, 2022-11-22:
still no luck.  I will  just purchase other type of MAX3232 breakout.

cole8888, 2022-11-22:
Darn.

Here is the one i use by the way: https://www.digikey.com/en/products/detail/mikroelektronika/MIKROE-602/4495610

cole8888, 2022-11-22:
Also which charge controller are you using?

Rey1635, 2022-11-22:
I'm using SRNE ML2420

cole8888, 2022-11-22:
Yeah that should work, appears to be nearly identical to the ML2440 which is one of the ones I use and the manual says it supports modbus and the bluetooth module which means we should be able to talk to it since the bluetooth module is pretty much doing the same thing. 

If you do get one of the ones with the DB9 connector (like the one I used) make sure you get a DB9 breakout adapter like [this](https://www.amazon.com/Connector-Adapter-Terminal-Signal-Module/dp/B071DS5GTW).

Rey1635, 2022-11-22:
ok. thanks for the suggestion, I'll do get that one.

Shadow-Genesis, 2022-11-22:
I know I maybe a little late on saying this. But to the serial board, I just found a USB cable for it. It is USB-A to RS232 Serial 6P6C on Amazon. I use a Raspberry PI for mine so USB works for me, and my work for others.

[https://www.amazon.com/dp/B07JGRJR4V?psc=1&ref=ppx_yo2ov_dt_b_product_details](url)

Rey1635, 2022-11-30:
Good day, I would like to ask if my connections are correct.
[image]
 and I changed the arduino code with this..
[image]

 Thank you very much

Rey1635, 2022-11-30:
Here's mine..
[image]

Rey1635, 2022-11-30:
changed the code
#define MAX3232_RX D1 // RX pin. If following my ESP8266 instructions use D1.
#define MAX3232_TX D2 // TX pin. If following my ESP8266 instructions use D2.

but  'D1' was not declared in this scope

cole8888, 2022-11-30:
Interesting, seems like some ESP8266 do not support the D labels. Lets just use the pin numbers directly.
```
#define MAX3232_RX 5 // RX pin. If following my ESP8266 instructions use D1.
#define MAX3232_TX 4 // TX pin. If following my ESP8266 instructions use D2.
```

cole8888, 2022-11-30:
Here is your diagram with the colors from your cable.
[image]

Rey1635, 2022-12-05:
Still no luck even with new arduino nano...  I followed your given pictures..
#define MAX3232_RX 2 
#define MAX3232_TX 3

cole8888, 2022-12-05:
The MAX3232 board has the same pinout as mine and the breakout should eliminate contact issues.

For the ground (Green on RJ12) are you using the GND terminal block on the breakout connector? If so can you switch to using pin 5 instead? It looks like some of these breakouts don't actually connect that terminal to pin 5.

Also just to clarify BLUE->3, YELLOW->2, GREEN->5

Rey1635, 2022-12-05:
I'm using pin 5 as ground on  breakout connector. also blue to pin 3, yellow to pin 2 and green to 5.  I will try reboot my charge controller

Rey1635, 2022-12-06:
Error when trying to communicate with the charge controller.  Here is my connections
[image]

cole8888, 2022-12-06:
Your connections look good. Let's try changing the ESP8266 pins, maybe those pins are reserved on your variant?

Move your Blue one (D1) over to D5 and Purple one (D2) over to D6 and then change the code to
```
#define MAX3232_RX 14
#define MAX3232_TX 12
```

cole8888, 2022-12-06:
As a sanity check I just went and did a fresh clone of the repo, changed only those two lines and using this configuration I was able to get data from my ML2440.
So I'm not really sure what's going on here. My money is on the MAX3232 not working correctly since the capacitors on that thing look way smaller than the ones on mine.

[image]

Rey1635, 2022-12-06:
Thank you.. I will replace my MAX3232 module

cole8888, 2022-12-06:
Hold on! I think I thought your connector was upside down, can you send another picture of your RJ12 jack?

cole8888, 2022-12-06:
One with the full connector please so I can see the wire colors through the clear plastic and see which way the wire exits the connector.

cole8888, 2022-12-06:
Like this:
[image]

Rey1635, 2022-12-06:
[image]

cole8888, 2022-12-06:
Yeah it's the way I thought it was, darn. I'm going to try the wiring diagram the renogy guides use where RX and TX are flipped relative to what I'm doing.
[image]

cole8888, 2022-12-06:
That didn't work, even if I flipped the ones on the ESP8266 side. Seems like the way I have it is the only way it works. I wonder if the renogy ones use a different pinout.

cole8888, 2022-12-06:
Good luck, I'm really sorry this is giving you so much trouble.

Rey1635, 2022-12-06:
Thats always part of learning.

Rey1635, 2022-12-08:
Good day, I've successfully read values from my charge controller and an ESP32 but using this code...
#define RXD2 16
#define TXD2 17

int num_data_registers = 30;
int num_info_registers = 17;

#include <ModbusMaster.h>
ModbusMaster node;

void setup()
{
  Serial.begin(115200);
  Serial.println("Started!");

  // create a second serial interface for modbus
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2); // is SERIAL_8N1 correct?

  int modbus_address = 255; // my Renogy Wanderer has an (slave) address of 255! Not in docs???
  node.begin(modbus_address, Serial2); 
}

void loop()
{
  static uint32_t i;
  uint8_t j, result;
  uint16_t data_registers[num_data_registers];
  uint16_t info_registers[num_info_registers];
  
  i++;
  
  // set word 0 of TX buffer to least-significant word of counter (bits 15..0)
  node.setTransmitBuffer(0, lowWord(i));  
  // set word 1 of TX buffer to most-significant word of counter (bits 31..16)
  node.setTransmitBuffer(1, highWord(i));
 
  // Read the 30 data registers
  result = node.readHoldingRegisters(0x100, num_data_registers);
  if (result == node.ku8MBSuccess)
  {
    Serial.println("Successfully read the data registers!");
    for (j = 0; j < num_data_registers; j++)
    {
      data_registers[j] = node.getResponseBuffer(j);
      Serial.println(data_registers[j]);
    }
    Serial.println("---");
  }
  else {
    Serial.print("Failed to read the data registers... ");
    Serial.println(result, HEX); // E2 is timeout
  }

  // Read the 17 info registers
  result = node.readHoldingRegisters(0x00A, num_info_registers);
  if (result == node.ku8MBSuccess)
  {
    Serial.println("Successfully read the info registers!");
    for (j = 0; j < num_info_registers; j++)
    {
      info_registers[j] = node.getResponseBuffer(j);
      Serial.println(info_registers[j]);
    }
    Serial.println("---");
    Serial.println();
  }
  else {
    Serial.print("Failed to read the info registers... ");
    Serial.println(result, HEX); // E2 is timeout
  }
  delay(1000);
}

Can I ask your help how to modify this code to read only the battery voltage and the charging state status? Thank you & God Bless

cole8888, 2022-12-08:
Awesome!
This is very interesting, can you tell me what modbus library you are using, also can you send a copy of the data you are receiving from the charge controller?
I do not have an ESP32 on hand so I won't be able to test it.
The modbus slave address is different than it is for all of my charge controllers, which is interesting. This could explain why you were not able to talk to it with my code.

Rey1635, 2022-12-08:
Library used is https://github.com/syvic/ModbusMaster.
Here is my serial output..
[image]
and the charge controller display..
[image]

Rey1635, 2022-12-08:
I found the code from here https://forum.arduino.cc/t/trying-to-read-registers-from-a-solar-charge-controller-over-rs232-solved/1039864/11

cole8888, 2022-12-08:
Hey can you include the data registers too? That's where the information you want is.

Rey1635, 2022-12-08:
Sorry, cant understand what data register is, I just copied the code from here https://forum.arduino.cc/t/trying-to-read-registers-from-a-solar-charge-controller-over-rs232-solved/1039864/11 and upload it to esp32

cole8888, 2022-12-08:
In the console output that script prints out two sets of data, one of them is called info registers and one is data registers. The data registers are the ones we want. There should be a header with the text "Successfully read the data registers!" printed above them.
Also could you send a picture of how you wired up the the ESP32? I'm trying to see if changing the modbus address will make my code work.

Rey1635, 2022-12-08:
console output for first set..
[image]

I just followed this..
[image]

cole8888, 2022-12-08:
Their TX and RX in the diagram seem to be opposite to mine so I just wanted to see how you physically hooked it up so I could assign the correct pin numbers in my code for you to try out.

cole8888, 2022-12-08:
An updated picture like this would be awesome: https://github.com/cole8888/SRNE-Solar-Charge-Controller-Monitor/issues/10#issuecomment-1338694262

Rey1635, 2022-12-08:
[image]

cole8888, 2022-12-08:
This data seems to be in the exact same format as what my program is expecting.

In the program we were working with previously can you try these settings?
`
#define MAX3232_RX 17
#define MAX3232_TX 16
`
and 
`
#define MODBUS_SLAVE_ADDR 255
`

Rey1635, 2022-12-08:
[image]

cole8888, 2022-12-08:
Shoot I got it backwards, try:
```
#define MAX3232_RX 16
#define MAX3232_TX 17
```
instead please.

Rey1635, 2022-12-08:
error SoftwareSerial.h: No such file or directory

[10 more replies]

closed 2023-01-09 by cole8888