cole8888/SRNE-Solar-Charge-Controller-Monitor #18: MobusIOException object has no attribute 'registers'

magorm, 2023-11-20:
Getting the error 'has no attribute registers' when running SolarMonitor.py. Need your help :-)

cole8888, 2023-11-21:
What charge controller are you using and how have you connected to it?

Can you try replacing the line containing `read_holding_registers` with `response = modbus.read_holding_registers(256, 35, unit=1, slave=255)`.

If your charge controller address is different than mine this will fix it.

magorm, 2023-11-21:
Thanks so much.
Here is how I connected SNRE SCC to my Laptop running the Python script:
[image]

magorm, 2023-11-22:
> What charge controller are you using and how have you connected to it?
> 
> Can you try replacing the line containing `read_holding_registers` with `response = modbus.read_holding_registers(256, 35, unit=1, slave=255)`.
> 
> If your charge controller address is different than mine this will fix it.

Did not work. I think the issue is related to USB adapter?

cole8888, 2023-11-22:
Which USB adapter are you using? It might not use the right voltage.

cole8888, 2023-11-22:
Also, can you show how you wired the serial adapter up to the RJ12 cable?

magorm, 2023-11-24:
> Which USB adapter are you using? It might not use the right voltage.

I am using this adapter: USB to TTL converter UART module CH340G CH340 3.3V 5V switch

magorm, 2023-11-24:
> Also, can you show how you wired the serial adapter up to the RJ12 cable?

[image]

cole8888, 2023-11-24:
The CH340G won't work, it only does +/-5V where RS232 needs +/-13V.

You'll need to use a usb to RS232 converter like this one:
https://www.amazon.ca/StarTech-com-USB-Serial-Adapter-Prolific/dp/B00GRP8EZU

Also, I can't see that image so I'm not sure how you wired it up to your charge controller.

magorm, 2023-11-25:
> The CH340G won't work, it only does +/-5V where RS232 needs +/-13V.
> 
> You'll need to use a usb to RS232 converter like this one: https://www.amazon.ca/StarTech-com-USB-Serial-Adapter-Prolific/dp/B00GRP8EZU
> 
> Also, I can't see that image so I'm not sure how you wired it up to your charge controller.

Thanks. 
Apologies. Here's the connection diagram.
[image]

magorm, 2023-11-25:
> The CH340G won't work, it only does +/-5V where RS232 needs +/-13V.
> 
> You'll need to use a usb to RS232 converter like this one: https://www.amazon.ca/StarTech-com-USB-Serial-Adapter-Prolific/dp/B00GRP8EZU
> 
> Also, I can't see that image so I'm not sure how you wired it up to your charge controller.

In your demo, you use Raspberry Pi, is the Pi serial using +-13v? Thanks in advance.

cole8888, 2023-11-25:
The Pi is not using that voltage. In my demo it has to go through a breakout board with a MAX3232 on it which handles the voltage boost to talk to the charge controller.

cole8888, 2023-11-25:
You can see it here:
https://github.com/cole8888/SRNE-Solar-Charge-Controller-Monitor/blob/main/RaspberryPi-Examples/RaspberryPi-Overview.jpg

magorm, 2023-11-25:
> MAX3232

Thanks. Will try this board, see if it will work - https://shopee.ph/MAX3232-RS232-to-TTL-Serial-Port-Converter-Module-DB9-Connector-MAX232-For-Arduino-i.226548939.4317808086

magorm, 2023-12-06:
Not yet. I already setup the raspberry pi zero 2w, made the code run but I think I bought a wrong RS232 adapter. I already ordered the board (https://www.digikey.ph/en/products/detail/mikroelektronika/MIKROE-602/4495610) similar to what you used. Hopefully I can make it work.

magorm, 2023-12-06:
[image]

magorm, 2023-12-08:
Hi. How did you get the register address? In the code its 256, not sure if it the same with SRNE charge controller?

magorm, 2023-12-09:
> I'm using SRNE charge controllers, which ones are you using?
> 
> I got it from the SRNE modbus datasheet

Thanks. I am using the same SCC.

cole8888, 2023-12-09:
I have that one also, so you definitely should be able to get it to work with it :)

magorm, 2023-12-09:
> I have that one also, so you definitely should be able to get it to work with it :)

In rj12, did you use pin1 (tx), pin2 (rx) and pin3(gnd)? pin4 is also gnd but not sure if I should use it instead of pin3?

cole8888, 2023-12-09:
It should work either way. I only used pin 3 for ground.

magorm, 2023-12-11:
Hi @cole8888 , for RJ12, the pin 1 is from the left side, right?

[image]

magorm, 2023-12-11:
This is the log when I run the script; it is sending  0xff 0x3 0x1 0x0 0x0 0x23 0x10 0x31?
Connection : True
DEBUG:pymodbus.logging:Current transaction state - IDLE
DEBUG:pymodbus.logging:Running transaction 1
DEBUG:pymodbus.logging:SEND: 0xff 0x3 0x1 0x0 0x0 0x23 0x10 0x31
DEBUG:pymodbus.logging:New Transaction state "SENDING"
WARNING:pymodbus.logging:Cleanup recv buffer before send: 0x0 0x0 0x0
DEBUG:pymodbus.logging:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.logging:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.logging:RECV: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
DEBUG:pymodbus.logging:Processing: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
DEBUG:pymodbus.logging:Getting transaction 255
DEBUG:pymodbus.logging:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Failed to read data, reconnecting... 'ModbusIOException' object has no attribute 'registers'

magorm, 2023-12-11:
For the DB9, its 2(rx), 3(tx) and 5(gnd)?

cole8888, 2023-12-11:
Black (controller rx) goes to pin 2, White (controller tx) goes to pin 3, red (controller gnd) goes to pin 5.

Remember that the pin numbers reverse when comparing the male/female connectors:
[image]

magorm, 2023-12-11:
Here's my connection. Not sure if the problem is the RJ12?

[image]

magorm, 2023-12-11:
[image]

magorm, 2023-12-11:
slave=255
[image]

magorm, 2023-12-11:
Serial Enablement: I am using /dev/ttyS0

[image]

magorm, 2023-12-12:
I was able to make it work! Thanks so much @cole8888 ! You are right the problem is wrong connection.

[image]

closed 2023-12-12 by cole8888