cole8888/SRNE-Solar-Charge-Controller-Monitor #19: Problems with the modbus

The-Cash, 2023-11-22:
I use the Raspberry Pi 3B+, after configuring and executing the script the following error message appears: 
`Failed to read data, reconnecting... 'ModbusIOException' object has no attribute 'registers'`
`Sent to topic ML4860`

I use the original USB converter PL2303 (but also a FTDI232 and a CH340). At least the PL2303 works on the computer with the manufacturer's software SolarStationMonitor. Do you have an idea what it could be?

The-Cash, 2023-11-22:
sorry read too late :)

'Can you try replacing the line containing read_holding_registers with response = modbus.read_holding_registers(256, 35, unit=1, slave=255).'

slave=255 was the solution

closed 2023-11-22 by The-Cash

cole8888, 2023-11-22:
Awesome! Did you have to change the serial port in the program? I think it's ttyS0 by default.

The-Cash, 2023-11-22:
slave=255 was the solution

I read it too late :)
Thanks for your great work! 👍