cole8888/SRNE-Solar-Charge-Controller-Monitor #20: Use with RS485?

korishan, 2024-09-22:
Would this project work with RS485 connections?  Would just replacing the RS232 with the RS485 adapter work?
I have a PowMr 5k Hybrid Inverter and right now I'm using the SNRE software to do the monitoring. Tho it is very basic and doesn't upload any information to MQTT or other storage.

I've been trying to get a working bridge to dump the data to influx and bring it into Home Assistant, but it's been trying times.

cole8888, 2024-09-22:
I'm not really sure this project would be best for that application, this is only really intended for the SRNE charge controllers and an inverter likely uses very different addresses for all the values you'd need. Also, I tried using RS485 with the charge controller but I wasn't able to get it to work, but the inverter may be different.

It looks like there are other projects specifically for powermr inverters like this one: https://github.com/leodesigner/powmr_comm

korishan, 2024-09-22:
Yeah, I've already tried using Leodesigner's code. I couldn't get it working. I don't know what the issue is.  Thanks for replying tho 👍

Zatarraz, 2025-09-15:
In my opinion the addresses are the same: I connected an elfin ew11a to the rs485 and, as a pure layman, I managed to obtain this data:

[image]

So i think that address and modbus are compatibile.